pbatard/libwdi #338: wdi-simple.exe does not install the driver

VolkovIA, 2024-12-11:
Hello, I compiled my project as written in the instructions here https://github.com/pbatard/libwdi/wiki/Compiling-and-debugging-libwdi-or-Zadig , but wdi-simple.exe unpacks the files and does nothing, outputs to the console - "driver files... Success" and "Installing the driver". After that, the console closes, and the device is still not detected correctly.

I tried to debug, but when I try to run the application in the debug, I get output to the console - 

"Extracting driver files...
libwdi:error [extract_binaries] Could not create file 'usb_driver\x86\WdfCoInstaller01011.dll' ([0x00000003] The system cannot find the path specified.)
  Could not allocate resource
C:\Users\User\libwdi\Win32\Debug\examples\wdi-simple.exe (process 32160) exited with code -3 (0xfffffffd)."

Zadig, however, works correctly.

VolkovIA, 2024-12-11:
> Please post the output from `wdi-simple -l 0`

[log.txt](https://github.com/user-attachments/files/18096469/log.txt)

pbatard, 2024-12-11:
I'm not seeing an error there...

VolkovIA, 2024-12-11:
Me too. But after the wdi-simple.exe, the device is displayed like this
[image]
And after zadig like this
[image]

Anyway, could you explain what the output - "libwdi:error [extract_binaries] Could not create file 'usb_driver\x86\WdfCoInstaller01011.dll' ([0x00000003] The system cannot find the path specified.)
Could not allocate resource" means during debugging?

pbatard, 2024-12-11:
It means that libwdi could not create `WdfCoInstaller01011.dll` in a `usb_driver\x86\` (which seems to be missing the full user directory path).

I'm afraid I cannot support custom builds of wdi-simple, so it's going to be up to you to use the debugger and find out why your wdi-simple is not working. The only executable I can support is Zadig. wdi-simple is provided as an example to build from and expects people to be familiar enough with application development (and troubleshooting) to build on it as needed.

You should also try to figure out what exactly is the error that Device Manager reports for your non-working driver, because, as far as libwdi is concerned, the driver is installed properly, and it looks like a device issue rather than a driver issue.

VolkovIA, 2024-12-11:
Ok, thanks!

closed 2024-12-11 by VolkovIA