pbatard/libwdi #341: Install driver only for a specific device

Benkol003, 2025-02-01:
This can be done by using `diInstallDevice`, and removing the `INSTALLFLAG_FORCE` flag from `UpdateDriverForPlugAndPlayDevicesU` in installer.c. Currently in zadig or with the installer, if there are multiple of the same device with the same hardware ID, then the driver will be replaced for all those devices.

Benkol003, 2025-02-01:
I think that the driver will still be installed for other matching devices if the driver has a higher rank. Maybe adding the driver to the driver store with `SetupCopyOEMInf` and removing `UpdateForPlugAndPlayDevicesU` to only use `DiInstallDevice` could fix this, but i have yet to test this out.

pbatard, 2025-02-01:
I'm afraid this is too specific of a request, and Zadig is designed for the general case where all devices with the same hardware ID should be served by the same driver.

If you want something different than this, I'm afraid you will have to sort you own driver installation rather than rely on an automated generic tool to do that for you.

closed 2025-02-01 by pbatard

Benkol003, 2025-02-01:
No worries :) i already have this implemented in a personal fork