pbatard/libwdi #347: zadig-2.9 一些使用技巧，som tips fo use zadig-2.9

unnow, 2025-03-11:
使用之前，如果你使用的windows系统，建议你使用高权限运行zadig-2.9，比如SYSTEM权限或者TrustedInstaller权限去运行，具体教程可以参考UP https://www.bilibili.com/video/BV1Ku4y1z7ic，对于TrustedInstaller权限使用请保持克制，不要运行你不信任的程序，使用高权限基本能解决50%以上的问题，如果使用高权限还是不能解决问题，大部分是因为zadig-2.9使用的默认目录为C导致的，切换高级模式，在C盘以外新建一个英文或者数字的文件夹解压文件，基本上90%以上的问题都能解决，毕竟zadig-2.9是替换系统相关USB的驱动程序。

AI translation
Before using, if you are using a Windows system, it is recommended to run zadig-2.9 with elevated permissions, such as SYSTEM or TrustedInstaller permissions. You can refer to the tutorial at UP https://www.bilibili.com/video/BV1Ku4y1z7ic. Please exercise caution when using TrustedInstaller permissions and do not run programs you do not trust. Using elevated permissions can resolve over 50% of issues. If problems persist even with elevated permissions, it is mostly due to zadig-2.9's default directory being C. Switch to advanced mode and create a folder with English letters or numbers outside of the C drive to extract the files, which can resolve over 90% of issues. After all, zadig-2.9 is used to replace system-related USB drivers.

[image]

unnow, 2025-03-11:
如果你觉得有用请点赞

If you find it useful, please like

pbatard, 2025-03-13:
1. Zadig automatically asks for elevation. It cannot run as non elevated. You should not need to run it as anything else but the default elevated user (Trusted Installer or System are not needed as the Windows driver installation API ultimate takes care of running the driver installation part as Trusted Installer). If you find that the defaults of Zadig regarding elevation do not work, then the problem is with your Windows configuration, as, on a vanilla installation of Windows, Zadig will ask for elevation, and the default elevated user will just work. I can of course not vouch for restricted or otherwise intentionally crippled installations of Windows, and I cannot support such platforms.
2. Again on a properly configured standard installation of Windows, the default directory to which Zadig copies its driver files is your user directory (Something like `C:\Users\<your user name>\`). If it defaults to `C:` then I would assume that your system has been configured/restricted in a nonstandard manner when it comes to user account, and there again, this is not something I can support. If you are experiencing such an issue, can you please indicate what `%USERPROFILE%` points to when you type in in File Explorer? Also, Zadig should support international characters for user directories, so I would certainly like to see what your user profile resolved to, to validate that there shouldn't be an issue there.

unnow, 2025-03-20:
感谢开发者回答我的问题，
1首先我是正常安装的Windows操作系统，右键使用管理员权限时，ziadig不能正常安装驱动，我没有保留当时报错LOG，下次安装驱动时我会上传上来
2 ziadig程序正确指向了我的用户路径（C:\Users\Administrator\usb_drive），但是这个目录是低权限程序不能删除和添加的，除非手动去点击授权，但是程序执行过程中不会有这个界面去手动点击授权，所以我才会在上面的建议中手动选择除C盘以外的位置，以保证程序正常运行。

Thank you for answering my question, 
1.first of all, I am a normal installation of the Windows operating system, when right-clicking to use the administrator permissions, ziadig can not install the driver normally, I did not keep the error LOG at that time, I will upload it the next time I install the driver
2.The program ziadig correctly points to my user path (C:\Users\Administrator\usb_drive), but this directory is one that low-privilege programs cannot delete or add to unless they are authorized manually. However, during the execution of the program, there is no interface to manually click for authorization, which is why I suggested manually choosing a location other than the C drive above to ensure the program runs normally.

unnow, 2025-03-24:
开发者您好

我重新安装了我的蓝牙驱动，全部使用管理员权限安装的，以下是没有更改安装目录的LOG,保存log时的报错图片，目录位置为：C:\Windows\System32\config\systemprofile\Desktop 我的电脑里没这个目录。
我在资源管理器中执行%USERPROFILE%指向了   C:\Users\Administrator\目录，但是查看了ziadig的log，默认的工作目录指向了C:\Windows\system32\config\systemprofile\，不知道为什么会默认保存到这里。

Hello developers
I reinstalled my bluetooth driver, all installed with administrator privileges, the following is the LOG without changing the installation directory, the error picture when saving the log, the directory location is: C:\Windows\System32\config\systemprofile\Desktop I don't have this directory in my computer.
I did %USERPROFILE% in Explorer and pointed to the C:\Users\Administrator\ directory, but I looked at ziadig's log, and the default working directory points to C:\Windows\system32\config\systemprofile\, and I don't know why it is saved here by default.

[Zadig no change directory.log](https://github.com/user-attachments/files/19433213/Zadig.no.change.directory.log)

[image]

这里是使用高级模式，更改安装目录以后的LOG，正常安装完成了。
Here is the LOG after using the advanced mode and changing the installation directory, and the normal installation is completed.

[Zadig change directory.log](https://github.com/user-attachments/files/19433214/Zadig.change.directory.log)

如果哪天我重新安装了操作系统后，使用管理员权限不能安装驱动的时候我会提供相关log，

If one day I reinstall the operating system and can't install the driver with administrator privileges, I will provide the relevant log

pbatard, 2025-03-24:
> when right-clicking to use the administrator permissions, ziadig can not install the driver normally

You shouldn't have to right click. Zadig should ask for elevation automatically. If your system doesn't do that, then I have to assume that something is nonstandard about your configuration, and I can not support that in Zadig.

> The program ziadig correctly points to my user path (C:\Users\Administrator\usb_drive)

If you are using `C:\Users\Administrator` then you are logging as `Administrator` rather than as a regular user. This is not a mode of operation I can support, as I can only support running Zadig as a regular user, which you are definitely not if Zadig points to `C:\Users\Administrator\usb_drive` for the path.

> the default working directory points to C:\Windows\system32\config\systemprofile, and I don't know why it is saved here by default.

Most likely this is again because you are not using a Windows regular user but the `Administrator`. I'm sorry, but this is not a configuration I can support, as a default installation of Windows will **not** use `Administrator` as a user, since this is reserved, but instead a different user name, that will be granted elevation privileges.

This means that I cannot help you further on this issue. There are just too many ways people can configure a Windows installation in a nonstandard way, that is going to break Zadig, and I just don't have the time to try to address these corner case scenarios.

Hopefully, you can reinstall Windows, and, when doing so, use a standard regular account for your user, and you should then find that Zadig works as expected.

closed 2025-03-24 by pbatard